Rubberazer/JETGPIO #3: Thank you for this library

GrahamBriggs, 2022-12-17:
This is not an issue, but a thank you.

I am new to working with Jetson, starting with some experience with the Raspberry Pi using wiringPi.

Thank you very much for creating and sharing this library. It was the only C library I could find that worked on both the NVIDIA developer carrier board and the Seeed Studio A203 carrier board I am using for my projects.

Thank you again

Graham Briggs

Rubberazer, 2022-12-20:
Many thanks for your feedback, this is greatly appreciated, having feedback from actual users is the most valuable thing when you do something like this library.
Comming back to your other comments/suggestions, I went through them and actually they mostly make sense to me. I plan to make some changes to my library in line with your suggestions and on the next release you will see those changes implemented. Most of them are very easy to do actually.
You will have to wait until mid/end January though as I am on a trip abroad and don't have my nano at hand, I wouldn' t like to release anything without some kind of testing on the actual hardware.
Thanks again

closed 2022-12-20 by Rubberazer

GrahamBriggs, 2022-12-20:
You are welcome. I am grateful to you for this work.  

Full disclosure:  when I started with Jetson what I really wanted to use was wiringPi, because I had quite a bit of useful code written using the wiringPi interface that I wanted to reuse on the jetson.

After some simple experiments with your library to verify it worked and that it could run out of the box on the Seeed Studio A203 carrier board I am using, I took a copy of your code and amalgamated it with Gordon Henderson's wiringPi code and created wiringJet. You may see it on GitHub here:  https://github.com/wiringGpio/wiringJet 

For your reference regarding the issues I entered

1) I did not use any of your code for the I2C communication. the I2C and SPI control in wiringJet is almost entirely verbatim from wiringPi.

2) I have implemented read from output pin and tested it on all pins. It appears to work. You can inspect that here: https://github.com/wiringGpio/wiringJet/blob/3c59fdcd937286980af9e5cc0d836a20e97ae333/wiringJet/jetgpio.c#L1168

3) I added a flag to disable the reset pins behavior in the terminate function:  https://github.com/wiringGpio/wiringJet/blob/3c59fdcd937286980af9e5cc0d836a20e97ae333/wiringJet/jetgpio.c#L962

I am not very experienced with low level stuff, and I don't really know a pinmux from a pin cushion, so I am going on faith and a bit of testing to verify my changes work correctly. Any guidance or feedback you can offer would be greatly appreciated.

Thank you again

GB

Rubberazer, 2022-12-20:
Cool, actually what made me create this lib was the fact that coming from the Raspberry pi universe I was used to the Pigpio library and all I wanted was to have something similar when using the Nano. I soon realized that there was no good options out there that just 'worked' and so I grabbed the Tegra CPU manual and started working on it on my spare time.
 
Anyways, I am still going to take note and do some improvements based on your comments plus something that is not there and it should be e.g. debouncing capabilities on the gpioSetISRFunc() function so the user can set up a debouncing time and so forth when catching rising/falling edges, once I am back home all this should take a very short time.

GrahamBriggs, 2022-12-20:
Sounds great. I hope you have safe and happy travels.
Best Regards
GB