Rubberazer/JETGPIO #4: I2C does not allow for multiple devices on the same bus

GrahamBriggs, 2022-12-17:
As a user, I need to have multiple I2C devices on the same bus.

In the function jetgpio.c / i2cOpen( )

This code (around line 1969) prevents more than one device on the same bus:

if (i2cInfo[i2cBus].state == I2C_CLOSED) {
		slot = i2cBus;
		i2cInfo[slot].state = I2C_RESERVED;
	}
	else { printf("i2c bus already open\n");
        return -4;
	}

Rubberazer, 2022-12-20:
You are absolutely right, I am going to make changes to my lib so all devices on the i2c bus can be addressed

Rubberazer, 2023-01-18:
Sorted, from current version (0.95) on, the I2C slave address has been removed from the i2cOpen() function and moved to the: i2cWriteByteData() & i2cReadByteData() functions so several devices can be reached on the same bus.

closed 2023-01-18 by Rubberazer