Rubberazer/JETGPIO #5: gpioRead read does not work with output pins

GrahamBriggs, 2022-12-17:
gpioRead( ) does not work when the pin is in output mode.

This may be by design, but it would be nice option if it could.  For what it is worth, the wiringPi library returns the state of the pin when you call digital read for both input and output mode pins. 

The code could look something like this:

```

int gpioRead(unsigned gpio)
{
	int level = 0;
	switch (gpio) 
	{
	case 3:
		if (*pinmux3 == PINMUX_IN)
			level = (pin3->IN[0]) >> 3 & 1;
		else
			level = (pin3->OUT[0]) >> 3 & 1;
		break;
	case 5:
		if (*pinmux5 == PINMUX_IN)
			level = (pin5->IN[0]) >> 2 & 1;
		else
			level = (pin5->OUT[0]) >> 2 & 1;
		break;
	case 7:
		if (*pinmux7 == PINMUX_IN)
			level = pin7->IN[0] & 1;
		else
			level = pin7->OUT[0] & 1;
		break;
	case 8:
		if (*pinmux8 == PINMUX_IN)
			level = pin8->IN[0] & 1;
		else
			level = pin8->OUT[0] & 1;
		break;
	case 10:
		if (*pinmux10 == PINMUX_IN)
			level = (pin10->IN[0]) >> 1 & 1;
		else
			level = (pin10->OUT[0]) >> 1 & 1;
		break;
```

Rubberazer, 2022-12-20:
It is a fair enough comment, will try to change this so all the pins set up as inputs can be read too

Rubberazer, 2023-01-18:
Sorted, from current version (0.95) on, function gpioRead() will give the actual level of a pin whether it has been setup as input or output with gpioSetMode()

closed 2023-01-18 by Rubberazer