Rubberazer/JETGPIO #6: gpioTerminate sets all pins back to their original state

GrahamBriggs, 2022-12-17:
gpioTerminate( ) will set all pins back to their original state.

I am sure this is by design. I suggest this is a bad design, because it means you can not reliably have more than one program using this library at the same time.

Consider this scenario

1. Program A starts, initializes the state of all pins
2. Program A is using pins 38 and 40, so sets those to desired mode
3. Program B starts, initializes state of all pins
4. Program B is using pins 29 and 30, so sets those pins to desired mode
5. Program A exits, it will reset pins 29 and 30 to their original state, which might not be desired state of program B, which is still running. Chaos will ensue.

Suggest that gpioTerminate should only reset the state of pins that have been initialized through call to pinMode in this instance of the program. Alternatively, do not reset pin mode state at all and make this a deliberate choice of the app using the library what to do with pins when it shuts down.

Rubberazer, 2022-12-20:
Yes this was a design decision but I see your point, I will think about it a bit and come up with something

Rubberazer, 2023-01-19:
From current version (0.96) on, only the pins used by the program(s) calling the library will be reverted back to their previous state and not all of them.

closed 2023-01-19 by Rubberazer