Rubberazer/JETGPIO #16: About PWM

ZiyaElri, 2023-12-12:
Hi,
 I have a Jetson Nano Developer Kit and I want to drive an SG90 servo motor (TowerPro Mİcro Serve 9g) using PWM. I can partially drive it. I downloaded and installed your library. However, I encountered an interesting issue. When I set the frequency to 50 Hz with the 'gpioSetPWMfrequency' function, I observe 45 Hz on the oscilloscope, and when I set it to 55 Hz, I see 50 Hz. What could be the reason for this? Have you encountered or seen any similar issues before? I am connecting the pins directly without using a resistor in between. I would greatly appreciate your help. Thank you in advance.

Rubberazer, 2023-12-12:
Hi there

You are right, the reason is that the clock divider that I had set up on my library was optimized to work on high frequencies but I hadn't bothered too much with the very low ones e.g. 50Hz, I overlooked it.
I corrected it, so now there are 2 different dividers depending on the frequency, one for small frequencies (less than 30kHz) and the other for the higher ones, from the usee point of view nothing to be done. 
Keep in mind that this hardware is not capable of producing PWM signals with absolute accurate frequencies, PWM it is not a clock.

To get the new version just clone + sudo make + sudo make install and let me know how it goes.

Related to the SG90, connections should be:

VCC to Nano pin2 (5VDC)
GND to Nano pin 6 (GND)
PWM input to Nano pin 32 or 33

I believe is going to work but the SG90 should get a PWM input at 5V, the Nano produces 3.3V so maybe you end up needing a level shifter, you will have to test.
Also think in terms of duty cycle, the SG90 moves from: 1ms (5% duty cycle at 50Hz), 1.5ms (7.5% duty cycle at 50 Hz) and finally 2ms (10% duty cycle at 50Hz). When you convert this to the lib (0-256) you get:

- Servo all the way to the right 5% -> 12.8 -> gpioPWM(32, 12)
- Servo centered 7.5% -> 19.2 -> gpioPWM(32, 19)
- Servo all the way to the left 10% -> 25.6 -> gpioPWM(32, 26)

Play around with those numbers and you should be fine.

ZiyaElri, 2023-12-16:
Hi dude,
Thank you for your response. I will install the updates on Monday and give it a try. I will share the results.

You mentioned the PWM values of SG90, and you are right, these are accurate. Between these values, SG90 operates between 0-180 degrees. 
Care yourself :)

Rubberazer, 2023-12-24:
Tested on devkit at: 40, 50, 60 and 100 Hz, all good

closed 2023-12-24 by Rubberazer