Rubberazer/JETGPIO #17: Qt and jetgpio_example.cpp problem

assyhgit, 2023-12-17:
hi, first of all thank you for pair for code..
my problem that, i am using Qt and i make copy code in jetgpio_example.cpp to mainwindow.cpp but editor show error 
 undefined reference to `gpioInitialise'
 undefined reference to `gpioSetMode'
 and other function, i did add in header #include <jetgpio.h> and read installation doc. make and make install, by the way , if i run jetgpio_example.cpp in with this way 
 Compile with: g++ -Wall -o jetgpio_example jetgpio_example.cpp -ljetgpio
 * Execute with: sudo ./jetgpio_example
 everything is okey. but in Qt doesnt work.
 have you got any idea..or solition
 

#include <stdint.h>
#include <stdlib.h>
#include <jetgpio.h>

MainWindow::MainWindow(QWidget *parent)
    : QMainWindow(parent)
    , ui(new Ui::MainWindow)
{

    this->setWindowFlags( (Qt::CustomizeWindowHint | Qt::WindowTitleHint | Qt::WindowMinMaxButtonsHint) & ~Qt::WindowMinimizeButtonHint );
    this->setFixedSize(QSize(pencere.en,pencere.boy));
    QMainWindow::setWindowFlags(Qt::FramelessWindowHint);
    ui->setupUi(this);
    int Init;

    Init = gpioInitialise();
    if (Init < 0)
    {
       /* jetgpio initialisation failed */
       printf("Jetgpio initialisation failed. Error code:  %d\n", Init);
       exit(Init);
    }
    else
    {
       /* jetgpio initialised okay*/
       printf("Jetgpio initialisation OK. Return code:  %d\n", Init);
    }

    // Setting up pin 3 as OUTPUT and 7 as INPUT

    int stat1 = gpioSetMode(3, JET_OUTPUT);
    if (stat1 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat1);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat1);
    }

    int stat2 = gpioSetMode(7, JET_INPUT);
    if (stat2 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat2);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat2);
MainWindow::MainWindow(QWidget *parent)
    : QMainWindow(parent)
    , ui(new Ui::MainWindow)
{

    this->setWindowFlags( (Qt::CustomizeWindowHint | Qt::WindowTitleHint | Qt::WindowMinMaxButtonsHint) & ~Qt::WindowMinimizeButtonHint );
    this->setFixedSize(QSize(pencere.en,pencere.boy));
    QMainWindow::setWindowFlags(Qt::FramelessWindowHint);
    ui->setupUi(this);
    int Init;

    Init = gpioInitialise();
    if (Init < 0)
    {
       /* jetgpio initialisation failed */
       printf("Jetgpio initialisation failed. Error code:  %d\n", Init);
       exit(Init);
    }
    else
    {
       /* jetgpio initialised okay*/
       printf("Jetgpio initialisation OK. Return code:  %d\n", Init);
    }

    // Setting up pin 3 as OUTPUT and 7 as INPUT

    int stat1 = gpioSetMode(3, JET_OUTPUT);
    if (stat1 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat1);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat1);
    }

    int stat2 = gpioSetMode(7, JET_INPUT);
    if (stat2 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat2);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat2);
    }
MainWindow::MainWindow(QWidget *parent)
    : QMainWindow(parent)
    , ui(new Ui::MainWindow)
{
    ui->setupUi(this);
    int Init;

    Init = gpioInitialise(); //**_<---- problem this line,**_
    if (Init < 0)
    {
       /* jetgpio initialisation failed */
       printf("Jetgpio initialisation failed. Error code:  %d\n", Init);
       exit(Init);
    }
    else
    {
       /* jetgpio initialised okay*/
       printf("Jetgpio initialisation OK. Return code:  %d\n", Init);
    }

    // Setting up pin 3 as OUTPUT and 7 as INPUT

    int stat1 = gpioSetMode(3, JET_OUTPUT);
    if (stat1 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat1);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat1);
    }

    int stat2 = gpioSetMode(7, JET_INPUT);
    if (stat2 < 0)
    {
       /* gpio setting up failed */
       printf("gpio setting up failed. Error code:  %d\n", stat2);
       exit(Init);
    }
    else
    {
       /* gpio setting up okay*/
       printf("gpio setting up okay. Return code:  %d\n", stat2);
    }
///-------------------------------------------------------------------///

 thank you...
assyhgit

Rubberazer, 2023-12-17:
Hi there

That "undefined reference" when you try to compile looks like my library is not being linked, if you look at my example:

g++ -Wall -o jetgpio_example jetgpio_example.cpp **-ljetgpio**

You notice that I have to tell that the jetgpio library has to be linked, not sure what your development environment is or if you are using make, cmake or something like that, but you have to make sure that my library is being linked, so for instance in the following Makefile (not a real project just to give you an example):

CC=g++
CFLAGS=-I. -I/usr/include/qt4/QtGui -I/usr/include/qt4/QtCore -I/usr/include/qt4
LIBS=-lQtGui -lQtCore **-ljetgpio**

app: main.cpp
    $(CC) $(CFLAGS) $(LIBS) $< -o $@

You notice how I put -ljetgpio within LIBS to make sure that is going to be linked

Rubberazer, 2023-12-18:
Related to how to do makefiles just google it, it is going to be easier for you and I am not an expert in the QT framework anyways.

Just to give you a brief explanation CFLAGS is a variable  the same as CC so for instance: CC CFLAGS -o jetgpio_example jetgpio_example.c -ljetgpio where CC = gcc and CFLAGS =  -Wall it is the same as writing: gcc -Wall -o jetgpio_example jetgpio_example.c -ljetgpio within the makefile.

Related to using sudo to execute any application created with my lib, there is nothing you can do about it, you have to use it. The error message is telling you why,  by using my lib you are reading/writing /dev/mem so you have to be root or use sudo, it is a protected file. A potential solution for not having to use sudo every time or log as root, would be to change the access rights/ownership of the /dev/mem file, I am going to leave it for you to google the stuff if you are interested!

assyhgit, 2023-12-19:
thank you for explain.you are right.  i will find a solition... may be run command on startup to myApp through .sh file , sudo eco etc. or uploader app 
thank you again..

closed 2023-12-19 by Rubberazer