Rubberazer/JETGPIO #18: segfault on GPIO configure

axibo-reiner, 2024-01-09:
Jetson Orin Nano
```
# R35 (release), REVISION: 3.1, GCID: 32827747, BOARD: t186ref, EABI: aarch64, DATE: Sun Mar 19 15:19:21 UTC 2023
```

The below error is caused when setting the GPIO to an OUTPUT. 

```
[ 1742.458497] ---[ end trace c9b4cbc2ab4c73c3 ]---
[ 1796.382888] CPU:0, Error: cbb-fabric@0x13a00000, irq=25
[ 1796.388290] **************************************
[ 1796.393217] CPU:0, Error:cbb-fabric, Errmon:2
[ 1796.397703] 	  Error Code		: FIREWALL_ERR

[ 1796.403372] 	  Error Code		: FIREWALL_ERR
[ 1796.407497] 	  MASTER_ID		: CCPLEX
[ 1796.410993] 	  Address		: 0x32e0000
[ 1796.414580] 	  Cache			: 0x0 -- Device Non-Bufferable
[ 1796.419771] 	  Protection		: 0x2 -- Unprivileged, Non-Secure, Data Access
[ 1796.426747] 	  Access_Type		: Read
[ 1796.430238] 	  Access_ID		: 0x15
[ 1796.430242] 	  Fabric		: cbb-fabric
[ 1796.437136] 	  Slave_Id		: 0x35
[ 1796.440356] 	  Burst_length		: 0x0
[ 1796.443845] 	  Burst_type		: 0x1
[ 1796.447157] 	  Beat_size		: 0x2
[ 1796.450373] 	  VQC			: 0x0
[ 1796.453149] 	  GRPSEC		: 0x7e
[ 1796.456195] 	  FALCONSEC		: 0x0
[ 1796.459425] 	**************************************
[ 1796.464467] ------------[ cut here ]------------
[ 1796.464486] WARNING: CPU: 0 PID: 0 at drivers/soc/tegra/cbb/tegra234-cbb.c:577 tegra234_cbb_isr+0x130/0x170
[ 1796.474494] Modules linked in: nvidia_modeset(OE) fuse(E) lzo_rle(E) lzo_compress(E) zram(E) ramoops(E) reed_solomon(E) loop(E) bnep(E) snd_soc_tegra186_asrc(E) snd_soc_tegra186_dspk(E) snd_soc_tegra210_ope(E) snd_soc_tegra210_mvc(E) snd_soc_tegra210_iqc(E) snd_soc_tegra186_arad(E) snd_soc_tegra210_afc(E) snd_soc_tegra210_dmic(E) snd_soc_tegra210_admaif(E) snd_soc_tegra210_adx(E) snd_soc_tegra210_amx(E) snd_soc_tegra_pcm(E) snd_soc_tegra210_mixer(E) snd_soc_tegra210_i2s(E) snd_soc_tegra210_sfc(E) aes_ce_blk(E) crypto_simd(E) cryptd(E) aes_ce_cipher(E) ghash_ce(E) rtl8822ce(E) r8168(E) sha2_ce(E) sha256_arm64(E) sha1_ce(E) rtk_btusb(E) btusb(E) snd_soc_tegra210_adsp(E) btrtl(E) btbcm(E) btintel(E) snd_soc_tegra_machine_driver(E) snd_soc_spdif_tx(E) snd_soc_tegra_utils(E) snd_hda_codec_hdmi(E) snd_soc_simple_card_utils(E) snd_soc_tegra210_ahub(E) r8169(E) realtek(E) userspace_alert(E) fusb301(E) tegra_bpmp_thermal(E) nvadsp(E) snd_hda_tegra(E) tegra210_adma(E) cfg80211(E) snd_hda_codec(E)
[ 1796.474603]  nv_imx219(E) snd_hda_core(E) spi_tegra114(E) nvidia(OE) binfmt_misc(E) ina3221(E) pwm_fan(E) nvgpu(E) nvmap(E) ip_tables(E) x_tables(E) [last unloaded: mtd]
[ 1796.474630] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G        W  OE     5.10.104-tegra #1
[ 1796.474632] Hardware name: NVIDIA Orin NX Developer Kit (DT)
[ 1796.474636] pstate: 60400089 (nZCv daIf +PAN -UAO -TCO BTYPE=--)
[ 1796.474638] pc : tegra234_cbb_isr+0x130/0x170
[ 1796.474640] lr : tegra234_cbb_isr+0x10c/0x170
[ 1796.474642] sp : ffff800010003e10
[ 1796.474643] x29: ffff800010003e10 x28: ffffd05082682680 
[ 1796.474646] x27: 0000000000000001 x26: 0000000000000080 
[ 1796.474649] x25: ffffd050820a7778 x24: ffffd050829ebe40 
[ 1796.474652] x23: ffffd05082387000 x22: 0000000000000019 
[ 1796.474655] x21: ffffd0508280ef20 x20: 0000000000000002 
[ 1796.474657] x19: ffffd0508280ef10 x18: 0000000000000010 
[ 1796.474660] x17: 0000000000000000 x16: ffffd050809e2d90 
[ 1796.474662] x15: ffffd05082682bf0 x14: ffffffffffffffff 
[ 1796.474665] x13: ffff800090003917 x12: ffff80001000391f 
[ 1796.474667] x11: 0101010101010101 x10: 7f7f7f7f7f7f7f7f 
[ 1796.474670] x9 : ffff800010003c30 x8 : 2a2a2a2a2a2a2a2a 
[ 1796.474672] x7 : 2a2a2a2a2a2a2a09 x6 : c0000000ffffefff 
[ 1796.474675] x5 : ffff7787a8377958 x4 : ffffd05082697968 
[ 1796.474677] x3 : 0000000000000001 x2 : ffffd05080b77460 
[ 1796.474679] x1 : ffffd05082682680 x0 : 0000000100010001 
[ 1796.474683] Call trace:
[ 1796.474685]  tegra234_cbb_isr+0x130/0x170
[ 1796.474694]  __handle_irq_event_percpu+0x60/0x2a0
[ 1796.474697]  handle_irq_event_percpu+0x3c/0xa0
[ 1796.474699]  handle_irq_event+0x4c/0xf0
[ 1796.474701]  handle_fasteoi_irq+0xbc/0x170
[ 1796.474705]  generic_handle_irq+0x3c/0x60
[ 1796.474707]  __handle_domain_irq+0x6c/0xc0
[ 1796.474712]  gic_handle_irq+0x64/0x130
[ 1796.474713]  el1_irq+0xd0/0x180
[ 1796.474720]  cpuidle_enter_state+0xb4/0x400
[ 1796.474722]  cpuidle_enter+0x3c/0x50
[ 1796.474726]  call_cpuidle+0x40/0x70
[ 1796.474727]  do_idle+0x1fc/0x260
[ 1796.474729]  cpu_startup_entry+0x28/0x70
[ 1796.474737]  rest_init+0xd8/0xe4
[ 1796.474744]  arch_call_rest_init+0x14/0x1c
[ 1796.474746]  start_kernel+0x50c/0x540
[ 1796.474748] ---[ end trace c9b4cbc2ab4c73c4 ]---
```

Rubberazer, 2024-01-09:
For context, if you can copy and paste your code please? If it is one of my examples just tell me which one.
This can take around a week for me to check, I am abroad and don't have my Orin at hand.

Rubberazer, 2024-01-15:
OK so I ran some examples on my Orin Nano here but I am not able to reproduce your problem, I can trigger outputs just fine. No cbb fabric "firewall" issues on my kernel log either.
Anyways, looking at your log in line "[ 1796.410993] 	  Address		: 0x32e0000"  this error is related to the Orin cbb firewall stopping an application in user space writing to register: 0x32e0000 which is the one controlling PWM7 aka PWM in pin 32. I tested  that also but I don't have that problem either (by installing my library that is taken care of)

So in short, with no more feedback from your side I am not able to find any problem on my side of things. There is an observation I would like to make though, your version of Linux for Tegra is: 35.3.1, that if I recall correctly was the first release supporting Orin Nano and that one was quite incomplete/buggy. Not sure what your actual problem is, but maybe reflashing with a newer version is a good idea, whether you use my library or something else, check the list here: https://developer.nvidia.com/embedded/jetpack-archive

Finally, ALL cbb-fabric firewall problems can be sorted out by modifying the device tree file controlling the behaviour of the cbb and then reflashing, for more info of how to disable the cbb firewall you can check this one: https://forums.developer.nvidia.com/t/i2c-reads-writes-over-32-bytes-cause-cbb-error-on-i2c-0/234855

closed 2024-01-17 by Rubberazer