Rubberazer/JETGPIO #19: cant run on boot

assyhgit, 2024-01-12:
hi,
i used your library , its wonderful and work but i have some diffucult for example
i run it in terminal with sudo ./myApp its worked but when i did with myapp.service its not running.  path is correct. permission is chmod +x myApp and i try chmod 777 myApp. can you try work on boot or do you have any idea.
i tried that, with desktop file, with service file, with .sh file . it not worked. but other app did works fine

thanks for your help
regards

Rubberazer, 2024-01-16:
This is all generic Linux stuff, so let's say you want your app to start on boot, a potential way of doing this could be to use Systemd, this below is an example:

- I compile one of my own examples:  jetgpio_output.c, which you can find in folder: EXAMPLES_C
- After compiling you will get an executable (application) called: jetgpio_output
- You then go and create a file with instructions for Systemd, let's call it: jetgpio_output.service
- You open the file and inside you will write instructions for Systemd (obviously you have to change the executable path to adjust for your own needs in ExecStart):
  [Unit]
Description=Starting jetgpio_output at boot

[Service]
ExecStart=/home/rubberazer/jetgpio/EXAMPLES_C/jetgpio_output

[Install]
WantedBy=multi-user.target

- You copy this file into: /etc/systemd/system
- After that you have to enable the service so it starts next time you boot: sudo systemctl enable jetgpio_output.service

That's it, your application will start automatically on boot, if you want to start it before booting for the sake of testing, you can type: sudo systemctl start jetgpio_output.service

It is worthy to mention that, sometimes, applications need to introduce some kind of delay waiting for some other services or whatever, you can do that, if needed just google how to do that kind of stuff on the .service file, it is not difficult

assyhgit, 2024-01-16:
thank you much.
i tried before myApp.service in /etc/systemd/system but ist not worked. than try this:

make StartUp Application comman in ubuntu menu to myApp.sh file and it must be with myApp in same folder
like this. /path/your/myAppsh.sh
myAppsh.sh:
#!/bin/bash
echo <your_pass> | sudo -S ./myApp

its worked.

but i will try again your said.
thank you so much,

closed 2024-01-16 by Rubberazer