Rubberazer/JETGPIO #22: Orin NX PWM minimum freq?

democat3457, 2024-04-08:
I'm running this on an Orin NX and I'm confused as to why the minimum PWM frequency is 400 Hz. As far as I'm aware, this locks me out of most servo and motor control (since those are usually 50-100Hz). Thanks!

Rubberazer, 2024-04-08:
Hi there

You are right, all those small servos don't tolerate going beyond 100Hz. 

I was aware that the minimum frequency running on the Orin pwms was a bit high, but at the time didn't give much importance to it, tested it, made sure it was working and moved on.

I updated the Orin source to take on frequencies from 50 to 1.59 GHz. it goes really nicely between 50 and 100kHz, from that point the wave distortion starts to be noticed and obviously becomes worse the higher it gets, but there is nothing I can do about it, that's the hardware.

So anyways, thanks for remind me about this, this pic shows pin 15 going nicely at 50Hz, have a go and let me know if anything odd.

[image]

democat3457, 2024-04-12:
Works perfectly! I updated the error print in #25 to reflect the new minimum Hz.

Rubberazer, 2024-04-12:
Closing then

closed 2024-04-12 by Rubberazer