Rubberazer/JETGPIO #31: spi max speed problem?

junny-kim, 2024-07-05:
I tried your code with the orin agx, 

but it seems that if I set the spi max speed to 500kHz, the communication works well but If I increase it higher than that, the communication becomes weird, is there some possible reason? I see that your original sample code sets the spi max speed to 5Mhz so it seems that higher spi speed is possible for you?

Rubberazer, 2024-07-10:
@junny-kim 

Check this other issue maybe there is a relation there: https://github.com/Rubberazer/JETGPIO/issues/33

At the moment, I suspect a driver issue, I can't help with that at this time

junny-kim, 2024-07-11:
thanks for your kind reply, 

agx orin works well with the spi loop test for me too with 50MHz maximum bus speed, but when connectred to the slave board the same prolblem still exists.
I also contacted the slave board company so if there is any further improvement I will let you know!

Thanks again for your time!

closed 2024-08-02 by Rubberazer