Rubberazer/JETGPIO #32: Run without sudo ?

avanmalleghem, 2024-07-08:
Hello, 

Do you have any ideas on how to run it without sudo ? This access to /dev/mem seems really a big deal...
Don't you have an alternative ? Like the way https://github.com/pjueon/JetsonGPIO do ?

Thanks a lot for your answer.

KR,

Antoine

Rubberazer, 2024-07-08:
As you are the second person in a week that is asking about running without sudo, I actually bothered and tried to find an alternative, obviously this has never been a problem to me so I never thought about it:

1. Use systemd and execute a script to start your app on boot or something like, you can find an example inside folder 'scripts' in my repo
2. Create a script with this inside (I don't like it but is easy): 
   echo your_user | sudo -S ./your_program
3. Do the following (I don't like it either):
    add your user to the kmem group: sudo usermod -a -G kmem your_user
    copy this file into /etc/udev/rules.d: sudo cp /lib/udev/rules.d/50-udev-default.rules /etc/udev/rules.d/
    edit this file above like:
             SUBSYSTEM=="mem", KERNEL=="mem|kmem|port", GROUP="kmem", MODE="0640"
             SUBSYSTEM=="mem", KERNEL=="mem|kmem|port", GROUP="kmem", MODE="0660"
   finally after compiling your app, change its capabilities: sudo setcap cap_sys_rawio+ep your_app
  you can now execute without sudo: ./your_app

3 is a pain in the neck, I am pretty sure if you spend time on google you will find some other options.

anath93, 2024-07-09:
@Rubberazer Same issue, using ROS2 with cmake side and can be tricky with hardware abstraction node running, any other alternative to these options using Nvidia approach ? (https://github.com/NVIDIA/jetson-gpio)

Rubberazer, 2024-07-09:
Not really, nothing straight forward. What Nvidia does in their lib is basically adding some new udev rules: lib/python/Jetson/GPIO/99-gpio.rules, that works with /dev/gpiochip0 and stuff like that but it doesn't with dev/mem because /dev/mem is actually special. Even if you become the owner of the file it doesn't matter because you will need to escalate the capabilities of your binary with setcap like above on top of everything else: sudo setcap cap_sys_rawio+ep your_app
You can't do that on a library, it has to be the actual executable (your app).

Now the reason to use /dev/mem is that: 1- is fast, 2-I can change settings like redefining a pin as input/output at runtime as many times as I want, you can't do that with the Nvidia lib, you have to modify the device tree every time.

Those 2 characteristics above can only be replicated on a kernel module, I could create a kernel module to do that and my library would wrap it up, but then again, the maintainers will never accept it upstream... so it would be like, what for?

I am really sorry but there is not much I can do. Just use the Nvidia lib if using sudo is an issue.

closed 2024-07-12 by Rubberazer