Rubberazer/JETGPIO #34: Reading issue on Jetson Orin Nano

leanedonze, 2024-07-17:
Hi Rubberazer,

First of all, thanks a lot for providing a library to use Jetson GPIOs in C, very helpful.

I have an issue reading an input on my Jetson Orin Nano. I don't think the issue is related to the library itself, but rather a configuration pin issue. However, I hope you might maybe give some advice.

When I press a button, I send a PWM (servo) signal to pin 15 of my Jetson Orin Nano. I am aware that the library has no function to read PWM, but I was able to write a custom code that worked with another system. I registered an ISR function that is supposed to be called and read the PWM. Currently, the ISR is never called. I made another test code using gpioRead() to print the pin level in the terminal and there is no change when I press the button. Finally, looking with an oscilloscope, I can see the PWM signal, but it is only 300mV at high state. When my remote controller is not connected to the Jetson, the PWM signal level is fine. 

I assume I need to configure the pin, because it seems to currently pull-down the line. Have you encountered a similar issue before?

Thank you,

Leane

Rubberazer, 2024-07-17:
So, what you are trying to do is monitor the pwm pins while producing a pwm signal in the Orin? I never tried, I can't even think what for.
OK, customer is always right I guess, I did a little change, clone an try again, to test it set up the pwm and use a gpioRead(), tell me how it goes.
Related to the weak output on the Orins, there is nothing I can do about it, you will need a buffer in many applications, is the hardware.

Rubberazer, 2024-07-17:
UPDATE:
Do not bother, it doesn't work I just tried, and actually I suspect it cannot be done on this machine, basically if the pin is doing PWM(SFIO) it can be not doing input(GPIO) at the same time. You can for instance read an output: gpioWrite(32, 1) & gpioRead(32) because both are in GPIO mode but this doesn't work when the same pin is in some SFIO e.g. PWM, SPI, UART...

Sorry about that, do not know how to do this one.

leanedonze, 2024-07-18:
Hi, 
Thank you for your reply.
I think my questions was not clear, sorry about that. All I want to do is read a PWM signal on a GPIO, no PWM generation from the Jetson. 
I send a PWM signal to the Jetson with my controller, but when I measure it, the signal is too weak and I cannot detect the edges. When the output of my controller is not connected to the Jetson, the signal is fine.
I think it could be a pull-down issue, but I am not sure at all.

Rubberazer, 2024-07-18:
Sorry about that,
Do they have common voltage reference? Are they powered from the same supply on the DC side? Do you measure 3.3 on the Orin GND and the other controller pin?

Just in case, have you tried to read on another pin?

leanedonze, 2024-07-18:
Thanks a lot for taking the time to help as the issue is not related to the library itself.

We tried pin 18 initially because this is the one used in the previous system. We encountered this issue, so we switched to pin 15, without any luck. But I might ask for some tests on other pins as well. 

I am not physically at the same place at the Jetson, but I will ask someone to check the voltage and connection and get back to you.

closed 2024-08-02 by Rubberazer