Rubberazer/JETGPIO #35: I2C communication bus speed path not found.

remilamoureuxlevesque, 2024-07-18:
I am having issue using the i2c communication on the Jetson Nano Orin. My jetpack version is 6.0-b52 and the os version is ubuntu 22.04.4 LTS. 

I am getting this error when I try to open the I2C bus : Not possible to read current bus speed. Segmentation fault.

By looking into the orin.c it seems that the file path for the bus speed file is no longer the same for with version I use. I will keep looking for the rigth path.

Thanks in advance for your help.

Rubberazer, 2024-07-18:
You are right, on the latest jetpack 6 the sysfs has changed, this is going to be interesting, lets see if i found another way or maybe i have to remove the option on jetpack 6 onwards

Rubberazer, 2024-07-18:
Not a solution to your problem but here it goes a rant.
So this is happening in other platforms as well, just checked some Odroid forum, is the Linux kernel people, nothing to do with Nvidia. Basically the only option open seems to be changing the device tree and reboot, no more dynamic changing. If there is no easy option I will have to disable the speed option for Jetpacks > 6.
By the way, the device tree is overused more and more and again the problem is that in principle the device tree doesn't allow changes at runtime. It can be done but only from a kernel module (double the work because then you will have to implement ioctls or something to be able to do stuff from user space).
Lets see if i can find some shortcut, if not the i2c will be functional but no more speed changes.

Rubberazer, 2024-07-18:
@remilamoureuxlevesque 
can you make me a favour and run this on your terminal? uname -r

remilamoureuxlevesque, 2024-07-19:
yes no problem, the output is :

5.15.122-tegra

Rubberazer, 2024-07-19:
Should be good now, no crashes. Flags will not have any effect for machines with Jetpack >= 6

remilamoureuxlevesque, 2024-07-19:
Thank for you responses, I update the library and now it works.

closed 2024-07-19 by remilamoureuxlevesque