Rubberazer/JETGPIO #36: `gpioInitialise()` Crashing System

yak-22, 2025-01-15:
Hi, I am trying to ingrate the library into my project and I was having issues. I condensed the problem into the below test program. 
When running the test program, the program goes unresponsive in the command line, it then crashes VS Code, and after several seconds the entire OS hangs and I am forced to restart. 
I'm wonder if it has to do with the jetpack version, although there is no mention of a minimum required version.

Test program:
```c++
#include <iostream>

#include <jetgpio.h>

int main(int argc, char* argv[]) {
    int init = gpioInitialise();
    
    if (init < 0)
        {
            /* jetgpio initialisation failed */
            printf("Jetgpio initialisation failed. Error code:  %d\n", init);
            exit(init);
    }

    gpioTerminate();    

    return 0;
}
```
This is the truncated `CMakeLists.txt`:
```cmake
include(FetchContent)
FetchContent_Declare(
    jetgpio
    GIT_REPOSITORY https://github.com/Rubberazer/JETGPIO.git
    GIT_TAG main
)

FetchContent_MakeAvailable(jetgpio)
file(GLOB JETGPIO_SOURCES "${jetgpio_SOURCE_DIR}/*.c")
file(GLOB_RECURSE JETGPIO_HEADERS "${jetgpio_SOURCE_DIR}/*.h")

add_library(jetgpio ${JETGPIO_SOURCES} ${JETGPIO_HEADERS})
include_directories(
    ${jetgpio_SOURCE_DIR}
    # ...
)
# ...
target_link_libraries(${PROJECT_NAME} PUBLIC jetgpio)
```
Jetpack information:
```bash
$ sudo apt-cache show nvidia-jetpack
    Package: nvidia-jetpack
    Version: 5.1.2-b104
    Architecture: arm64
    Maintainer: NVIDIA Corporation
    Installed-Size: 194
    Depends: nvidia-jetpack-runtime (= 5.1.2-b104), nvidia-jetpack-dev (= 5.1.2-b104)
    Homepage: http://developer.nvidia.com/jetson
    Priority: standard
    Section: metapackages
    Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.1.2-b104_arm64.deb
    Size: 29304
    SHA256: fda2eed24747319ccd9fee9a8548c0e5dd52812363877ebe90e223b5a6e7e827
    SHA1: 78c7d9e02490f96f8fbd5a091c8bef280b03ae84
    MD5sum: 6be522b5542ab2af5dcf62837b34a5f0
    Description: NVIDIA Jetpack Meta Package
    Description-md5: ad1462289bdbc54909ae109d1d32c0a8
```
OS Version: `Ubuntu 20.04.6 LTS`
Hardware: `Jetson Orin NX 16GB`

Rubberazer, 2025-01-15:
I believe the problem is trying to automate the dependency on your Cmakelists. Just download the repo directly on your Jetson NX and compile/install the old fashion way e.g.:

          sudo make   
          sudo make install  

After that , just compile your own example (or any other in the repo). It should be fine.
On the reason why you are experiencing this, during compilation my makefile  compiles&runs a little program ([get_chip_id.c](https://github.com/Rubberazer/JETGPIO/blob/main/get_chip_id.c))  that determines automatically your Jetson model and whether is supported or not, by doing what you are doing, I believe that CMake is just compiling the wrong version (for the old Nano for instance), that indeed would crash your system as the library will be trying to read a set of registers that don't correspond to that Jetson model.

Try that and if everything works fine, remove the dependency solve/fetch from your lists (or figure out how to make it work knowing what I just explained, it should be doable).

yak-22, 2025-01-16:
Ok, thank you.
I'll try that and report the results.

yak-22, 2025-01-16:
I installed via 
```bash
sudo make   
sudo make install  
``` 
and I can confirm that it does indeed work. 

I initially had it with `FetchContent_Declare` so that it would compile on Windows even if it wouldn't actually run on Windows. I might still do that with a CMake Windows target. I figure something out.

Regardless, thanks again for making this library and for quick and through answer.

Rubberazer, 2025-01-17:
Cool,

Just to help you a bit with your CMakeLists, as you already know your model (Orin NX) you can skip the autodetection altogether on my Makefile by doing this:

               sudo make orin

and then of course install. It will compile directly to a binary compatible with your Jetson model.
So knowing this, it should be simpler to prepare your fetch, the rest is up to you though :)

closed 2025-01-19 by Rubberazer