Rubberazer/JETGPIO #38: System Crash on Init

a-zane, 2025-04-22:
Thanks for your work in creating and maintaining this library!

I've got an issue that seems quite similar to [this recent issue](https://github.com/Rubberazer/JETGPIO/issues/36), but a manual installation doesn't seem to solve the problem. The chip detection also seems to be working correctly in identifying my Orin AGX.

This program causes my system to freeze and reboot some minutes later.
```
#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <unistd.h>
#include <jetgpio.h>

int main(int argc, char *argv[])
{
  int Init;

  Init = gpioInitialise();
  if (Init < 0)
    {
      /* jetgpio initialisation failed */
      printf("Jetgpio initialisation failed. Error code:  %d\n", Init);
      exit(Init);
    }
  else
    {
      /* jetgpio initialised okay*/
      printf("Jetgpio initialisation OK. Return code:  %d\n", Init);
    }
  // Terminating library 
  gpioTerminate();

  exit(0);
}
```

Here's output of my `sudo make`
```
gcc -Wall -Werror -o get_chip_id get_chip_id.c
./get_chip_id
T234/Orin AGX detected
gcc -c -Wall -Werror -fpic orinagx.c
gcc -shared -o libjetgpio.so orinagx.o -lpthread -lrt -lm
```

And of the subsequent `sudo make install`
```
install -m 0755 libjetgpio.so /usr/lib
install -m 0644 jetgpio.h /usr/include
ldconfig
ldconfig -p | grep libjetgpio.so
	libjetgpio.so (libc6,AArch64) => /lib/libjetgpio.so
make[1]: Entering directory '/home/user/JETGPIO/Jetclocks'
make -C /lib/modules/5.10.65-tegra/build M=/home/user/JETGPIO/Jetclocks modules
make[2]: Entering directory '/usr/src/linux-headers-5.10.65-tegra-ubuntu20.04_aarch64/kernel-5.10'
make[2]: Leaving directory '/usr/src/linux-headers-5.10.65-tegra-ubuntu20.04_aarch64/kernel-5.10'
make[1]: Leaving directory '/home/user/JETGPIO/Jetclocks'
make[1]: Entering directory '/home/user/JETGPIO/Jetclocks'
make[2]: Entering directory '/usr/src/linux-headers-5.10.65-tegra-ubuntu20.04_aarch64/kernel-5.10'
  INSTALL /home/user/JETGPIO/Jetclocks/jetclocks.ko
  DEPMOD  5.10.65-tegra
Warning: modules_install: missing 'System.map' file. Skipping depmod.
make[2]: Leaving directory '/usr/src/linux-headers-5.10.65-tegra-ubuntu20.04_aarch64/kernel-5.10'
jetclocks.conf created into /etc/modules-load.d/
Finding device tree blob
Creating backup of: /boot/dtb/kernel_tegra234-p3701-0000-p3737-0000.dtb -> /boot/kernel_tegra234-p3701-0000-p3737-0000.dtb.jetclocks.backup
Applying overlay blob to: /boot/dtb/kernel_tegra234-p3701-0000-p3737-0000.dtb
Creating backup of: /boot/extlinux/extlinux.conf -> /boot/extlinux/extlinux.conf.jetclocks.backup
Modifying: /boot/extlinux/extlinux.conf
All done, you should reboot your machine now
make[1]: Leaving directory '/home/user/JETGPIO/Jetclocks'
```

And here's my install from `sudo apt-cache show nvidia-jetpack`. I'm currently on an older jetpack version, which might be part of the problem.
```
Package: nvidia-jetpack
Version: 5.0.1-b118
Architecture: arm64
Maintainer: NVIDIA Corporation
Installed-Size: 194
Depends: nvidia-jetpack-runtime (= 5.0.1-b118), nvidia-jetpack-dev (= 5.0.1-b118)
Homepage: http://developer.nvidia.com/jetson
Priority: standard
Section: metapackages
Filename: pool/main/n/nvidia-jetpack/nvidia-jetpack_5.0.1-b118_arm64.deb
Size: 29304
SHA256: 78f182886cc85b6467f37e32553ee79adcf4aaf6b68031fe948f33cc8e008f92
SHA1: 3371de1f109f313379ed19f41a860fb60a555c86
MD5sum: 51cf5d858d2e533f063f3a7467bed6da
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8

Package: nvidia-jetpack
Status: install ok installed
Priority: standard
Section: metapackages
Installed-Size: 194
Maintainer: NVIDIA Corporation
Architecture: arm64
Version: 5.0.1-b118
Depends: nvidia-cuda (= 5.0.1-b118), nvidia-opencv (= 5.0.1-b118), nvidia-cudnn8 (= 5.0.1-b118), nvidia-tensorrt (= 5.0.1-b118), nvidia-container (= 5.0.1-b118), nvidia-vpi (= 5.0.1-b118), nvidia-nsight-sys (= 5.0.1-b118), nvidia-l4t-jetson-multimedia-api (>> 34.1-0), nvidia-l4t-jetson-multimedia-api (<< 34.2-0)
Description: NVIDIA Jetpack Meta Package
Description-md5: ad1462289bdbc54909ae109d1d32c0a8
Homepage: http://developer.nvidia.com/jetson
```

Any insight would be much appreciated!

Rubberazer, 2025-04-23:
Hi there

A quick one, what is your machine brand and model? Meaning, is an official Nvidia dev kit? Seed studio, Waveshare or other? If you can tell me exactly which one it would be helpful.

a-zane, 2025-04-23:
It's the [official NVIDIA dev kit](https://www.arrow.com/en/products/945-13730-0057-000/nvidia?utm_source=google&utm_medium=cpc&utm_campaign=brand_g-pmax-amer_Arrow_en-PerformanceMax-Development%20Kits_q1_2025&utm_content=na_en&gclsrc=aw.ds&gad_source=1&gclid=EAIaIQobChMIrfy5p8fujAMVmSZECB3YBT1mEAQYAyABEgJKXPD_BwE). Let me see if I can get a part number...

Rubberazer, 2025-04-23:
With an official dev kit or a clone it should not be happening, weird stuff. Can you try to comment: gpioTerminate() like:

//gpioTerminate();

Compile that example and try again?

a-zane, 2025-04-23:
Bottomed out with the following, but I didn't get any print statements. I'm not used to working this close to the metal so I'm not sure if that's strange.

```c
/* Usage example of the JETGPIO library
 * Compile with: gcc -Wall -o jetgpio_init jetgpio_init.c -ljetgpio
 * Execute with: sudo ./jetgpio_init
 */

#include <stdio.h>
#include <stdint.h>
#include <stdlib.h>
#include <unistd.h>
#include <jetgpio.h>

int main(int argc, char *argv[])
{
  int Init;

  printf("About to initialize");

  Init = gpioInitialise();
  if (Init < 0)
    {
      /* jetgpio initialisation failed */
      printf("Jetgpio initialisation failed. Error code:  %d\n", Init);
      exit(Init);
    }
  else
    {
      /* jetgpio initialised okay*/
      printf("Jetgpio initialisation OK. Return code:  %d\n", Init);
    }
  // Terminating library 
  // gpioTerminate();

  exit(0);
}
```

a-zane, 2025-04-23:
Is there a way to easily check whether the gpio register locations are where we think they are?

Rubberazer, 2025-04-23:
Registers should be fine, I tested it myself back in the day and people have been using it on AGX since June 2024. Unfortunately I can't replicate on my side at the moment because I toasted my AGX in January.

It is behaving like if it was a different machine, so basically what happens sometimes is that if you have a board different to the dev kit (or a clone) even if the module is the same e.g. Orin AGX, as the pins are routed differently, when you mmap() the registers you might read/write a memory block already taken by a driver and therefore you might have a kernel crash and a reboot. Say module pin E60 goes to pin3 on dev kit, but in some other board it could go to a USB or whatever.

What you can do if you want, is try the older version of the library, Not high hopes but is low effort, link here:
https://github.com/Rubberazer/JETGPIO/archive/refs/tags/v1.2.zip

Please first uninstall your current version with: sudo make uninstall

To install the older version is the same: sudo make && sudo make install

Let me know how it goes if so, I am really curious/mildly upset, I hate when things fail.

a-zane, 2025-04-23:
Got it. Is that just a hardware thing or can kernel updates change how those memory blocks are allocated?

Doing some more research, here are some possible culprits. Wondering if these ring any bells for you?

```$ zcat /proc/config.gz | grep STRICT_DEVMEM
CONFIG_STRICT_DEVMEM=y
# CONFIG_IO_STRICT_DEVMEM is not set```
```$ cat /proc/cmdline
root=/dev/mmcblk0p1 rw rootwait rootfstype=ext4 mminit_loglevel=4 console=ttyTCU0,115200 console=tty0 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0```

I've seen some posts that resolve related issues by adding `iomem=relaxed` to `/boot/extlinux/extlinux.conf`(so that it shows up in that second line).

Rubberazer, 2025-04-23:
No, if /dev/mem is not reachable you will get an error and that little test program would stop, actually the [get_chip_id.c](https://github.com/Rubberazer/JETGPIO/blob/main/get_chip_id.c) program to detect the module type would fail too.

a-zane, 2025-04-23:
What kit and kernel version did you do your testing on?

Rubberazer, 2025-04-23:
1. Latest version was 5.15.x if I recall correctly, but I have never experienced those symptoms in any Jetson because kernel version, only wrong register configuration/wrong board or the like
2. Nothing that I personally experienced

Sorry but as I am not able to replicate on my side, I am kind of stuck, maybe you can try other options. I am going to leave this open in case other people is experiencing something similar.

a-zane, 2025-04-23:
Alright, so I had another Orin AGX on hand, and everything seems to be working with that one! I was able to run the round trip speed test, and am getting under 4.2 microseconds as expected. This second AGX is basically new out of the box, so there must be some "aftermarket" stuff breaking the first AGX.

Off the top of my head, the first one has had a few different camera drivers added, a driver for a barcode scanner, and perhaps some others I'm not thinking about. Any ideas on where to start playing "spot the difference"?

Rubberazer, 2025-04-23:
Glad to hear, it was puzzling me, I was giving up.
Again, I never experienced those symptoms personally (a reboot) due some other software that I installed or such but potentially anything kernel occupying those memory segments could do it. 
Something that you can definitely do is check your kernel logs with dmesg + journalctl, for the relevant registers that my lib uses for the Orin AGX check jetgpio.h, there are comments.

Finally, something that I haven't mentioned but these machines incorporate the Arm Trust zone technology, so basically they have sort of a "register firewall" you have an MCU sort of controlling what the CPU cores (the ones running Linux) are doing to peripherals, it is configurable during flashing at least and modifiable. That could potentially cause something like that, but I have never seen it, usually they cause kernel panics, but those are not dangerous. This can be disabled if you know how to, but documentation is not great.
Interesting if you are say, Tesla, Roomba or whatever and you want to limit other peoples intervention in your software for instance, anti hacking measures basically.
I explored it back in the day and is definitely something one can play with, achieved a couple of things but then I stopped because no clear application on my side, interesting stuff.

closed 2025-04-25 by Rubberazer