Rubberazer/JETGPIO #42: Help for AGX Xavier

zenchf, 2025-06-18:
Has anyone tried and used the lib on an agx xavier device? can anyone help ? 
I tried so many times but i can just light a led with pin 8

zenchf, 2025-06-18:
> Hi there
> 
> My library does NOT support the Xavier AGX, only the NX model.
> 
> If one day, somebody donates a Xavier AGX so I can develop and test, then the AGX will be also supported.

I have an agx xavier, but I could not control gpio with C++ and file operations, I have been working on it for a few weeks. If you help me, we can make your library compatible with agx xavier. What exactly needs to be done for agx xavier? Determine the pins? How do we adapt this??

Rubberazer, 2025-06-18:
Sorry but I dont work remote, e.g. me developing and somebody else testing all the pins somewhere else. Unless I have the machine in my hands I won't update the library to support the Xavier AGX.

I don't have time to work remote with other people, is slow and not very efficient, and quite honestly with my day job I have enough. I work for free on this library on my spare time.

Sorry, but you will have to find some other solution,

oznurkarahasan, 2025-06-18:
> Sorry but I dont work remote, e.g. me developing and somebody else testing all the pins somewhere else. Unless I have the machine in my hands I won't update the library to support the Xavier AGX.
>
> I [...]

Okay, thank you for your time. Actually I understand you and you are right but we are developing a project and we are students. we cannot solve this problem and we need help. At least give us an idea of what we can do. We feel so helpless.

Rubberazer, 2025-06-18:
Start by articles like this: https://forums.developer.nvidia.com/t/xavier-nx-gpio-character-device-interface/267961

Then click on other references within that one, I reckon it will take you sometime but you should be able to arrive somewhere.

Sorry, but there is no easy recipe with these machines, the reason is having the right setup on the device tree.

You can also try to emulate what I did, I explain the mechanics of it in a discussion in my repo, this one:
https://github.com/Rubberazer/JETGPIO/discussions/10#discussioncomment-11543477
It is not for beginners though.